stunt822/SteadyPass #2: PID overshoots and oscillates

stunt822, 2018-05-03:
Under some circumbstatces - control will overshoot and oscillate speed.  For example when speed is below target - throttle increase may be too slow, so speed is dropping further.  Leading to oscillation

Test comment. But real from one of users.

CRaShCoUrSe316, 2018-05-07:
* Original calculation by MPH has been removed and replaced with PID Calculation
* PID Needs testing/calibration and further adjustments on the water.

!* RPM mode needs to be replaced

CRaShCoUrSe316, 2018-05-08:
*Combined RPM and speed mode with a single function that changes the input when the mode is changed.
*Removed previous calculation style.
*Two modes of calculation available and set by an EPPROM reading (read EEPROM(19) for a 1 to set adaptive tuning mode, is not ==1 then Proportion on Measure mode is set by default.

CRaShCoUrSe316, 2018-05-08:
TODO
* integrate mode change into menu option
*integrate calc type into menu option
*enhancement- add knob turn setting to adjust the aggressiveness of the PID calculation for a harder/softer launch

CRaShCoUrSe316, 2018-05-24:
New PID functioning, Needs testing and adjustments on Kp,Ki,Kd settings

CRaShCoUrSe316, 2018-06-04:
New PID has been water tested on initial No pull at 11.5 mph. At best, held a range of 11-12 mph. some outbreaks to wider range during testing. Settings need to be dialed in much better to properly gauge accuracy. More testing needed to see if issue is resolved with new PID calculations.

CRaShCoUrSe316, 2018-08-21:
Target speed holding perfectly with current setting. Overshoot still needs to be fixed.

closed 2018-08-21 by CRaShCoUrSe316