stunt822/SteadyPass #26: Code stepper motor support (alternative FW to linear servo)

CRaShCoUrSe316, 2018-06-04:


Birdy718, 2021-11-02:
Hi, just wondering if this ever got any traction? I have a steady pass unit that I bought last year but unfortunately I was fitting this to a 200hp outboard and the servo Dimitri supplied isn't strong enough to pull back the throttle. I knew this was a possibility comming in as I spoke to him about it. Now that we are comming into the season again in Australia im keen to get this working but I am not a programmer. I am happy to throw some money at this though for someone if they think this could be a possibility.  Looking to run a 12v servo as it would be much more powerful than a 5v pwm

Sticky-Wicket, 2021-11-02:
Do you have any pictures on how you installed the servo from the kit?

Birdy718, 2021-11-02:
[image]
i do not have any pictures of my set up, but i did the same as this. When i tested the setup it was able to let off the throttle but was unable to pull it back on. was not strong enough as these outboards have there triggering system also on the throttle.

Birdy718, 2021-11-02:
Alternative if you know where i can purchase a 12v stepper that supports PWM that might have enough grunt to pull it back.

Birdy718, 2021-11-17:
any thoughts?

Sticky-Wicket, 2021-11-28:
Do you have any way to determine how much force is needed to pull your throttle? I am assuming you may be able to rig a digital fish scale to a string (or something like that) to measure this.  You will need to disconnect your throttle input when measuring this as we only want to know what it takes to pull on the throttle not actually trying to move your whole throttle lever.

Birdy718, 2021-11-28:
I can do that sometime this week. i was hoping to use the same servo as Perfect Pass does as i have seen these installed on the same motor I have work fine. But obviously looking for a cheaper alternative to Perfect Pass

Birdy718, 2021-12-16:
https://m.aliexpress.com/item/32918288522.html

Maybe this is a solution?